polarfire-soc/meta-polarfire-soc-yocto-bsp #3: issues building hss

codusnocturnus, 2020-09-15:
Ran into several errors in the hss recipe trying to build on WSL2.

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-poky-linux"
MACHINE              = "icicle-kit-es"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.2"
TUNE_FEATURES        = "riscv64"
meta
meta-poky
meta-yocto-bsp       = "dunfell:87c9d296357638082f5a7131a87365285fc813c4"
meta-oe
meta-python
meta-multimedia
meta-networking
meta-riscv           = "<unknown>:<unknown>"
meta-polarfire-soc-yocto-bsp = "master:5097cab102860ab43a1cc44d64ca68a49852ad56"

```

```
ERROR: hss-1.0+gitAUTOINC+589b29c29e-r0 do_configure: Execution of '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/temp/run.do_configure.5445' failed with exit code 1:
python interpreter details: sys.version_info(major=3, minor=8, micro=2, releaselevel='final', serial=0)
python interpreter running is version 3
output header files created in hardware/ directory
mv: cannot move '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/git/hardware' to '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/git/boards/icicle-kit-es/config/hardware': No such file or directory
WARNING: exit code 1 from a shell command.
```

```
ERROR: hss-1.0+gitAUTOINC+589b29c29e-r0 do_configure: Execution of '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/temp/run.do_configure.26123' failed with exit code 1:
python interpreter details: sys.version_info(major=3, minor=8, micro=2, releaselevel='final', serial=0)
python interpreter running is version 3
output header files created in hardware/ directory
ln: target '/path/to/build/tmp-glibc/hosttools/' is not a directory: No such file or directory
WARNING: exit code 1 from a shell command.
```

```
ERROR: hss-1.0+gitAUTOINC+589b29c29e-r0 do_configure: Execution of '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/temp/run.do_configure.26745' failed with exit code 1:
python interpreter details: sys.version_info(major=3, minor=8, micro=2, releaselevel='final', serial=0)
python interpreter running is version 3
output header files created in hardware/ directory
ln: failed to create symbolic link '/path/to/build/tmp/hosttools/genconfig': File exists
WARNING: exit code 1 from a shell command.

```

```
ERROR: hss-1.0+gitAUTOINC+589b29c29e-r0 do_install: Execution of '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/temp/run.do_install.28218' failed with exit code 1:
install: cannot stat '/path/to/build/tmp/work/icicle_kit_es-poky-linux/hss/1.0+gitAUTOINC+589b29c29e-r0/git/hss.*': No such file or directory
WARNING: exit code 1 from a shell command.

```

Cyril-Jean, 2020-09-17:
I think I know what happened. The Hart Software Services (HSS) was updated to use FPGA configuration settings using a slightly different folder structure. This project's recipe for the HSS does not appear to specify the hash for the HSS to use.
A wild guess is that modifying the HSS recipe in meta-polarfire-soc-yocto-bsp/recipes-bsp/hss/hss.bb will resolve the issue.

I am not a yocto expert but adding the SRCREV with the hash below might fix the issue:
SRC_URI = "git://github.com/polarfire-soc/hart-software-services.git;branch=master"
SRCREV = "9b620330ca67456fdcad643f9ae98bb785a09740"

codusnocturnus, 2020-09-17:
I guess it depends on whether/when you want to pull those changes into the Yocto build?

Changing the recipe to use a fixed revision will keep such structural changes in the HSS project from breaking the Yocto build, but those changes were presumably made on purpose, and should be pulled in here at some point?

I submitted a PR that addresses the structural changes to HSS (as of 9/16, at least), so you could merge that and change the SRCREV to 589b29c29eb5f44492c0c7d37a28b95428a5b743, if it's desirable to use the latest version of HSS.

Cyril-Jean, 2020-09-18:
The recipe was changed to build with HSS 0.99.9. I am a lot more comfortable with specifying the hash of the HSS to build with to prevent similar problems in the future.
There is one more change set queued up. Can you hold on until it shows up before making any changes to your pull request?

codusnocturnus, 2020-10-07:
OBE as of 9/28

closed 2020-10-07 by codusnocturnus