polarfire-soc/meta-polarfire-soc-yocto-bsp #6: Runqemu does not work

Le0nRoy, 2020-10-14:
Hello!

Trying to launch qemu for image builded by 

> $  MACHINE=icicle-kit-es-sd bitbake -k mpfs-dev-cli

and other combinations of MACHINE / image recipe, but all times getting next output.

> yocto@b18bbb4e826a:~/input/build$ ./../openembedded-core/scripts/runqemu
> runqemu - INFO - Running bitbake -e ...
> ls: cannot access '/home/yocto/input/build/tmp-glibc/deploy/images/icicle-kit-es-sd/*.qemuboot.conf': No such file or directory                                                                                       
> runqemu - ERROR - Command 'ls -t /home/yocto/input/build/tmp-glibc/deploy/images/icicle-kit-es-sd/*.qemuboot.conf' returned non-zero exit status 2.                                                                   
> runqemu - INFO - Cleaning up

Should I do some additional step to get .qemoboot.conf file?

Thank you for your time!

eppidei, 2020-10-14:
Hi @Le0nRoy  you have to build 
$MACHINE=qemuriscv64 bitbake  mpfs-dev-cli
and then
$runqemu qemuriscv64

lewishanly, 2020-10-15:
As stated above works.
$MACHINE=qemuriscv64 bitbake {image} 
$runqemu qemuriscv64

lewishanly, 2020-10-15:
MACHINE=qemuriscv64 ./openembedded-core/scripts/runqemu nographic

Le0nRoy, 2020-10-15:
> As stated above works.
> $MACHINE=qemuriscv64 bitbake {image}
> $runqemu qemuriscv64

Thank you much, it helped!

closed 2020-10-15 by Le0nRoy