polarfire-soc/meta-polarfire-soc-yocto-bsp #19: PCIe space limited so no space for modern PCIe graphics cards

bjdooks-ct, 2021-03-11:
The PCIe low space is too small to accept a 2GiB window for even a modest modern PCIe graphics card.

mcnamarad1971, 2021-05-20:
Just seeing this now.  This issue is firmly on our list of things to do.  No firm release date yet for this.

bjdooks-ct, 2021-09-21:
Hi, I just installed the HSS/OpenSBI from 2021.08 and the yocto from master and it is still not detecting the PCIe graphics card.

bjdooks-ct, 2021-09-21:
For reference, the graphics card we are trying is a HD6450 2G DDR3

bjdooks-ct, 2021-09-21:
Adding dmesg output from system with graphics card.
[new1cap.txt](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/files/7204814/new1cap.txt)

jjardon, 2021-10-06:
This is being discussed here as well: https://exchange.riscv.org/t/pcie-device-not-enumerated/85/14

hughbreslin, 2023-03-08:
Hi all, please test with the latest release [2023.02](https://github.com/orgs/polarfire-soc/discussions/297) as this has resolved a number of issues in relation to supported cards with our PCIe implementation.

closed 2026-03-06 by vfalanis