polarfire-soc/meta-polarfire-soc-yocto-bsp #21: reboot command hangs on ICICLE-KIT-ES

ant9000, 2021-04-05:
Issuing a reboot command stops the YOCTO image, but the system does not restart and periodically shows

```
rcu: INFO: rcu_sched self-detected stall on CPU
```

debug messages. A complete log is attached.

[reboot.log](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/files/6261005/reboot.log)

lhanlyu, 2021-04-14:
This is currently not support. Currently in development and after the next opensbi update it will be integrated.

dilyn-corner, 2022-08-18:
Is there an update on this? I haven't built a Yocto image myself, but the image my kit came with also has issues with this.

hughbreslin, 2022-08-19:
Hi @dilyn-corner, this is mentioned in [Discussion 101](https://github.com/orgs/polarfire-soc/discussions/101) its not implemented yet and we're working on it. The plan is to have support in our September (.09) Linux release

dilyn-corner, 2022-08-19:
Thanks for letting me know!