polarfire-soc/meta-polarfire-soc-yocto-bsp #22: MMC Card initialisation errors

pathakraul, 2021-04-08:
Hi,

I have flashed core-image-minimal-dev-icicle-kit-es-20210216165805.rootfs.wic.gz, in EMMC, flashing was successful but after the subsequent reboot getting the errors in mmc initialisation and boot process gets stuck after the that.

Error Logs
```
[    1.147603] Waiting for root device /dev/mmcblk0p3...
[    1.336222] mmc0: error -110 whilst initialising MMC card
[    1.608298] mmc0: error -110 whilst initialising MMC card
[    1.889277] mmc0: error -110 whilst initialising MMC card
[    2.197884] mmc0: error -110 whilst initialising MMC card
[   64.519189] random: fast init done

```

Full Error Logs
```
Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Dec 23 2019, 02:06:26.
Port /dev/ttyUSB1, 08:03:54

Press CTRL-A Z for help on special keys

[11.586544] HSS_Boot_PMPSetupHandler(): Hart1 setup complete
HSS_OpenSBI_Setup(): MTVEC switching from 20220230 to 20220100

U-Boot 2020.10 (Feb 16 2021 - 16:13:33 +0000)

CPU:   rv64imafdc
Model: Microchip MPFS Icicle Kit
DRAM:  1 GiB
MMC:   mmc@20008000: 0
In:    serial@20100000
Out:   serial@20100000
Err:   serial@20100000
Net:   eth0: ethernet@20112000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr.uimg
1103 bytes read in 10 ms (107.4 KiB/s)
## Executing script at 88100000
4170800 bytes read in 357 ms (11.1 MiB/s)
## Copying 'fdt@microchip_icicle-kit-es-a000-microchip.dtb' subimage from FIT image at 88300000 ...
sha256+    Loading part 253 ... OK
## Loading kernel from FIT Image at 88300000 ...
   Using 'conf@microchip_icicle-kit-es-a000-microchip.dtb' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x883000fc
     Data Size:    4159588 Bytes = 4 MiB
     Architecture: RISC-V
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    sha256
     Hash value:   50d73173d0d97470de8e75c77dd39a699902e3f49b19093c7cdc40c479b5dfc1
   Verifying Hash Integrity ... sha256+ OK
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Uncompressing Kernel Image
   Using Device Tree in place at 0000000088000000, end 00000000880053d5

Starting kernel ...

[... 97 lines cut ...]
[    1.039414] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 0 active
[    1.046198] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 1 active
[    1.052953] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 2 active
[    1.059755] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 3 active
[    1.066560] pac193x 0-0010: :pac193x_prep_iio_channels: Active chip channels: 25
[    1.084131] NET: Registered protocol family 10
[    1.090950] Segment Routing with IPv6
[    1.094874] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    1.101727] NET: Registered protocol family 17
[    1.107101] mpfs-sys-controller soc:syscontroller: Registered MPFS system controller driver
[    1.116135] mpfs-serial-number soc:serialnum: Successfully registered mpfs serial number driver
[    1.125345] mpfs-digest soc:fpgadigest: Successfully registered mpfs fpga digest driver
[    1.134835] mpfs-rng soc:hwrandom: Successfully registered HW RNG
[    1.141046] hctosys: unable to open rtc device (rtc0)
[    1.147603] Waiting for root device /dev/mmcblk0p3...
[    1.336222] mmc0: error -110 whilst initialising MMC card
[    1.608298] mmc0: error -110 whilst initialising MMC card
[    1.889277] mmc0: error -110 whilst initialising MMC card
[    2.197884] mmc0: error -110 whilst initialising MMC card
[   64.519189] random: fast init done

```

Thanks
Rahul

RafaelLaya, 2021-04-09:
Have you updated to the latest HSS 0.99.15? We also found this issue until we updated to this version

pathakraul, 2021-04-09:
No, I haven't. Thanks for pointing this.

lhanlyu, 2021-04-14:
Update to the latest design 2021.2 (includes HSS)
https://github.com/polarfire-soc/icicle-kit-reference-design

pathakraul, 2021-04-20:
Are there consolidated steps anywhere to just update just the HSS? Please point me to that.

lhanlyu, 2021-04-20:
Please see the following:
https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/software-development/polarfire-soc-software-tool-flow.md#using-the-hss

RafaelLaya, 2021-04-22:
Also pathakraul, be aware that after updating you might be able to boot but it is still unstable. If you're developing something then just try many times until you get lucky (e.g.: I was able to boot yesterday and it hasn't crashed in 24hrs, but most of the times it doesn't boot or crashes after a few seconds). But obv. think very careful about what you push into production

closed 2021-08-18 by lhanlyu