polarfire-soc/meta-polarfire-soc-yocto-bsp #23: payload.bin is not used from the boot partition

eenurkka, 2021-04-09:
I've taken the latest official yocto build and I've flashed it to the device with the windows tool. When I mount the eMMC via the USBDMSC -command from the HSS prompt, and update the contents (fitImage, payload.bin and boot.scr.uimg) I notice that the /proc/version shows correct date so the kernel (fitImage) is okay. However, my payload.bin contains "Apr 29" timestamp, but on the command line I see the tag: "U-Boot 2020.10 (Feb 16 2021 - 16:13:33 +0000)". Thus, I conclude the payload.bin is not used from the boot partition? 

I want to use my own binary in place of payload.bin, which I'd like to update to the eMMC. How can I manage it since the payload.bin at the boot partition appears unused?

griffini, 2021-04-14:
The payload.bin appears to have been unintentionally included in the boot partition - that copy is not used and it will be removed.

The HSS is able to parse the GPT partition table to find a special partition, but it does not have a filesystem driver. It looks for a BIOS Boot Partition (https://en.wikipedia.org/wiki/BIOS_boot_partition) and pulls the payload.bin directly from that sector.   Yocto generates one of these partitions and programs the U-Boot to it (typically as the second partition) in the .wic file it generates.

To update the payload.bin, you will need to overwrite this portion.  When you use USBDMSC, you'll see Linux mounts the boot and root partitions, but the BIOS boot partition is unmounted.

Let's assume Linux has recognised the eMMC as /dev/sdc, and that partition /dev/sdc2 is the BIOS boot partition (i.e., has a GUID of 21686148-6449-6E6F-744E-656564454649).  You can then do something like ...

````
# sudo dd if=payload.bin of=/dev/sdc2 bs=40960 status=progress
````

... to reprogram the boot payload.

In Linux, you can also the loopback device to mount a .wic directly, and reprogram the BIOS boot partition in it. See https://www.linux.org/docs/man8/losetup.html for more details.

lhanlyu, 2021-04-24:
Payload.bin is removed  from boot partition in this release.

closed 2021-08-18 by lhanlyu