polarfire-soc/meta-polarfire-soc-yocto-bsp #24: HowTo -  Boot Icicle-Kit Qemu machine with prebuilt yocto image

pathakraul, 2021-05-30:
Hi, 

I have been following the steps to boot the  microchip-icicle-kit machine on qemu from its documentation - https://qemu-project.gitlab.io/qemu/system/riscv/microchip-icicle-kit.html
I built the latest HSS on my own and downloaded the latest Icicle Linux image from github.
HSS - 2021.04
Linux Image - 20210428094631

Upon executing the qemu, it just waits for the connection on the `unix:serial1` and even if I connect on the unix:serial1 using `sudo minicom -D unix\#serial1.sock`, nothing happens on both serial consoles. 

```
rpathak@~/P/w/o/microchip> qemu-system-riscv64 -M microchip-icicle-kit -smp 5 -bios hart-software-services/Default/hss.bin -sd core-image-minimal-dev-icicle-kit-es-2021042809
4631.rootfs.wic -nic user,model=cadence_gem -nic tap,ifname=tap,model=cadence_gem,script=no -display none -serial stdio -chardev socket,id=serial1,path=serial1.sock,server=on
,wait=on -serial chardev:serial1                                                                                                                                              
qemu-system-riscv64: -chardev socket,id=serial1,path=serial1.sock,server=on,wait=on: info: QEMU waiting for connection on: disconnected:unix:serial1.sock,server=on
```

Am I missing something here? I have not tried building the Yocto yet and will do that but is the Qemu documentation obsolete and does not work anymore?

Thanks
Rahul

pathakraul, 2021-06-02:
I solved the problem and it seems that this qemu machine (microchip-icicle-kit)  has nothing to do with polarfire yocto, i realised that later. So closing the issue

Shivaraj199, 2022-04-26:
@pathakraul Can you please provide me steps how did you emulate polarfire on Qemu?

pathakraul, 2022-04-28:
The steps i followed were same as described in qemu microchip-icicle-kit machine documentation https://qemu.readthedocs.io/en/latest/system/riscv/microchip-icicle-kit.html 

The problem i was facing was because of the sockets which were not working on my system, so I used `pty` as chardev

Its been a while and i dont have the command right now

Shivaraj199, 2022-05-08:
How did you get the img as the in the Qemu page the link is dead!