polarfire-soc/meta-polarfire-soc-yocto-bsp #25: PCIe SSD performance is not even as good as the eMMC

taovcu, 2021-05-30:
I'm using the latest 2021.4 release on the Polarfire SoC.

I attached a Intel PCIe SSD to the board and did a performance tests. The IOPS for 1KB IO requests is capped at around 9500. On a x86, it can achieve 400k+. 

In comparison, the RISC-V eMMC can achieve an IOPS of 11000+. 

I'm curious about why the PCIe SSD cannot beat the eMMC. Does anyone have any idea? Thanks.

mcnamarad1971, 2021-05-31:
We're trying to make sense of PCIe performance in general at the moment.   We expect to be able to make improvements but the exact timeline is to be determined yet.

taovcu, 2021-06-01:
[image]
[image]

I did comprehensive tests of the PCIe SSD performance, and compared the device performance on Polarfire with an x86 server. The first figure reports the SSD access bandwidth with various request sizes for sequential/random read and write. The second figrue reports the SSD IOPS with 1KB request size, using libaio of various queue depth. My observation is that the PCIe SSD access is CPU intensive (I see the Polarfire CPU utilization is always nearly 100%), on x86 the CPU utilization is ~30% to achieve 82k IOPS of 4KB random write. 

I ran "openssl speed" to compare the CPU core performance of Polarfire U54 core with the x86 Intel Xeon 4215 core. I see the U54 core performance is about 3% of the Xeon 4215 core. Therefore, my conclusion is that currently the PCIe SSD performance on Polarfire is limited by the U54 application core's relatively low computing power. 

Why the eMMC can performance better than the PCIe SSD, my observation is that the eMMC access is not CPU intensive, maybe because the IO stacked is well optimized. 

This is my understanding. To optimize the PCIe SSD performance on the Polarfire, a key is to review the PCIe access IO stack and figure out what component consumes considerable CPU cycles and optimize it. 

I just played with the board for a few weeks, some observations maybe inaccurate. I'll continue investigate and update/correct here. Thanks.

mcnamarad1971, 2021-06-08:
please do.  i'll also be working on this for the next few weeks

hughbreslin, 2023-03-08:
Hi all, please test with the latest release [2023.02](https://github.com/orgs/polarfire-soc/discussions/297) as this has resolved a number of issues in relation to supported cards with our PCIe implementation.