polarfire-soc/meta-polarfire-soc-yocto-bsp #27: mpfs missing kernel patch and does not build.

sdanfa, 2021-06-16:
Hi,
The following patch file is missing `file://v11-0003-PCI-microchip-Add-host-driver-for-Microchip-PCIe.patch`
I could not successfully compile for mpfs because of that missing patch file. Also is there a xfce4 build recipe?

Thanks.

lhanlyu, 2021-06-17:
Updates will be in next release for mpfs support, build was broken in last couple of releases. We do not have image for desktop. Debian has being ported for the RISCV and you could use that image.

sdanfa, 2021-06-17:
> Updates will be in next release for mpfs support, build was broken in last couple of releases. We do not have image for desktop. Debian has being ported for the RISCV and you could use that image.

Okay I see thanks. Do you have a kernel image and a dtbs I could test with the debian rootfs? I tried older fu-sdk from sifive but I'm getting a init failed error.

sdanfa, 2021-06-17:
Can I use buildroot kernel with custom linux rootfs like debian instead? Since yocto don't build.

sdanfa, 2021-06-18:
I posted on buildroot github also: https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk/issues/6

sdanfa, 2021-08-10:
@lhanlyu @Cyril-Jean Does the new yocto build supports mpfs with expansion board? Thanks.

Cyril-Jean, 2021-08-10:
No, it does not. The Hi-Five + Expansion board support will be in a different branch.

sdanfa, 2021-08-10:
> No, it does not. The Hi-Five + Expansion board support will be in a different branch.

Okay thanks for your reply. Do you have any ETA when that will happen?

Cyril-Jean, 2021-08-11:
@sdanfa The branch __mpfs-dev-kit__ targeting the HiFive Unleased + Expansion Board is now available from this repo.

sdanfa, 2021-08-11:
> @sdanfa The branch **mpfs-dev-kit** targeting the HiFive Unleased + Expansion Board is now available from this repo.

Thanks. I'm building it right now.

sdanfa, 2021-08-11:
Nvm it tries tftp then boots via image. Can you tell me if this output means expansion board is working? I don't get any video output. Here's what it says

`root@mpfs:~# cut -f1,2,18 /proc/bus/pci/devices
0000	11aa1556	pcieport
0100	11f88561	pcieport
0200	11f88561	pcieport
0208	11f88561	pcieport
0210	11f88561	pcieport
0218	11f88561	pcieport
0300	11063483	xhci_hcd
0400	10026778	
0401	1002aa98	
0600	1b4b9125	ahci
`

sdanfa, 2021-08-11:
I'm using AMD Radeon HD6450. But there's no video output.

lhanlyu, 2021-08-12:
Looks like pcie is working, console is not redirected out video port. I would not expect to see any video with the current images.
You would need to add a desktop, like xfce desktop which we do not have at the moment.

sdanfa, 2021-08-12:
@Cyril-Jean @lhanlyu I try to change CONFIG_CMDLINE to add "rw" command to it but changes are not being made. I change the file "meta-polarfire-soc-yocto-bsp/recipes-kernel/linux/files/mpfs/extra.cfg" to include "rw" in CONFIG_CMDLINE but when I rebuild the kernel via "bitbake virtual/kernel" the changes don't reflect in fitImage.

sdanfa, 2021-08-12:
I got radeon to load via .ipk files converted to deb pkgs for my debian rootfs.

lhanlyu, 2021-08-13:
Good news, is all working for you OK?

sdanfa, 2021-08-13:
> Good news, is all working for you OK?

I think it's working as I'm getting amd Radeon related outputs on dmesg but I think my graphic card is dead since it does not output anything. Xorg server doesn't give errors though that's why I think my amd card is dead.

lhanlyu, 2021-08-18:
Seperate branch for mpfs board (Hifive Board with expansion) and lc-mpfs.
mpfs-dev-kit

closed 2021-08-18 by lhanlyu