polarfire-soc/meta-polarfire-soc-yocto-bsp #29: Extensible SDK build fails

codusnocturnus, 2021-09-21:
Following the instructions in #13, the basic SDK builds, but the extensible SDK fails with the following error messages:

ERROR: Task mpfs-amp-examples.do_fetch attempted to execute unexpectedly
Task /.../icicle-kit/build/tmp-glibc/work/icicle_kit_es-oe-linux/core-image-minimal/1.0-r0/sdk-ext/image/tmp-renamed-sdk/layers/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:do_image_qa, unihash e7d066010bfcbe6b0165a68c9a664689c2df7729bbc0ee24ff13698c9c24d2ac, taskhash e7d066010bfcbe6b0165a68c9a664689c2df7729bbc0ee24ff13698c9c24d2ac
Task /.../icicle-kit/build/tmp-glibc/work/icicle_kit_es-oe-linux/core-image-minimal/1.0-r0/sdk-ext/image/tmp-renamed-sdk/layers/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:do_image_complete, unihash 387eef45e8ba79e90132ae47f461d2087eda860c7151c45b19c2a3b5c717d166, taskhash 387eef45e8ba79e90132ae47f461d2087eda860c7151c45b19c2a3b5c717d166
This is usually due to missing setscene tasks. Those missing in this build were: {'/.../icicle-kit/build/tmp-glibc/work/icicle_kit_es-oe-linux/core-image-minimal/1.0-r0/sdk-ext/image/tmp-renamed-sdk/layers/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:do_image_complete',
 '/.../icicle-kit/build/tmp-glibc/work/icicle_kit_es-oe-linux/core-image-minimal/1.0-r0/sdk-ext/image/tmp-renamed-sdk/layers/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:do_image_qa'}
ERROR: Task (/.../icicle-kit/build/tmp-glibc/work/icicle_kit_es-oe-linux/core-image-minimal/1.0-r0/sdk-ext/image/tmp-renamed-sdk/layers/openembedded-core/meta-polarfire-soc-yocto-bsp/recipes-amp/mpfs-amp-examples/mpfs-amp-examples.bb:do_fetch) failed with exit code 'setscene whitelist'

lhanlyu, 2022-05-05:
We are in the process of fixing this and it will be available in the next release.

lhanlyu, 2022-05-27:
Fixed in release 2022.05

closed 2022-05-27 by lhanlyu