polarfire-soc/meta-polarfire-soc-yocto-bsp #32: Only 1 out of the 4 CPU cores made available to Linux

nearly-big-endian, 2021-12-13:
Hi,

I am currently running the [2021.08](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/tree/v2021.08) release on a icicle-kit-es, and only one of the 4 cpu cores could be made available at boot.
Do you have any idea why ?

I also tested the board with a [Buildroot based system](https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk), just to make sure it wasn't defective, and the four cores are usable.

Some commands on the Yocto build:
Notice the errors about CPU1,2,3 failing to come online in the dmesg output.

```
# dmesg | grep -i -e SMP -e cpu
[    0.000000] Linux version 5.12.19 (oe-user@oe-host) (riscv64-oe-linux-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP Thu Nov 4 13:00:56 UTC 2021
[    0.000000] CPU with hartid=0 is not available
[    0.000000] CPU with hartid=0 is not available
[    0.000000] percpu: Embedded 17 pages/cpu s30744 r8192 d30696 u69632
[    0.000000] pcpu-alloc: s30744 r8192 d30696 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=5 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] CPU with hartid=0 is not available
[    0.000000] riscv-intc: unable to find hart id for /cpus/cpu@0/interrupt-controller
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [2]
[    0.064317] smp: Bringing up secondary CPUs ...
[    1.140953] CPU1: failed to come online
[    2.182231] CPU2: failed to come online
[    3.223872] CPU3: failed to come online
[    3.227954] smp: Brought up 1 node, 1 CPU

# cat /proc/cpuinfo
processor       : 0
hart            : 2
isa             : rv64imafdc
mmu             : sv39
uarch           : sifive,u54-mc
cpu MHz         : 0.00

#  ls -l /sys/devices/system/cpu/
drwxr-xr-x    5 root     root             0 Aug 24 11:04 cpu0
drwxr-xr-x    3 root     root             0 Aug 24 11:04 cpu1
drwxr-xr-x    3 root     root             0 Aug 24 11:04 cpu2
drwxr-xr-x    3 root     root             0 Aug 24 11:04 cpu3
drwxr-xr-x    2 root     root             0 Aug 24 11:06 hotplug
-r--r--r--    1 root     root          4096 Aug 24 12:50 isolated
-r--r--r--    1 root     root          4096 Aug 24 12:50 kernel_max
-r--r--r--    1 root     root          4096 Aug 24 11:05 offline
-r--r--r--    1 root     root          4096 Aug 24 11:05 online
-r--r--r--    1 root     root          4096 Aug 24 11:05 possible
-r--r--r--    1 root     root          4096 Aug 24 12:50 present
drwxr-xr-x    2 root     root             0 Aug 24 12:50 smt
-rw-r--r--    1 root     root          4096 Aug 24 11:04 uevent

# cat /sys/devices/system/cpu/online
0
# cat /sys/devices/system/cpu/offline
1-3
# cat /sys/devices/system/cpu/possible
0-3
# cat /sys/devices/system/cpu/present
0-3
# cat /sys/devices/system/cpu/kernel_max
4

# echo 221 >/sys/devices/system/cpu/cpu1/hotplug/target
[ 6749.180898] CPU1: failed to come online
sh: write error: Input/output error
```

Same set of commands on the Buildroot build (same target):
Notice how the 4 cores are all enabled & usable by default.

```
# dmesg | grep -i -e SMP -e cpu
[    0.000000] Linux version 5.12.1-00063-gbae76cfb297a (root@Ubuntu-Test) (riscv64-unknown-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.36.1) #3 SMP Thu Nov 11 10:50:53 CET 2021
[    0.000000] CPU with hartid=0 is not available
[    0.000000] CPU with hartid=0 is not available
[    0.000000] percpu: Embedded 17 pages/cpu s32680 r8192 d28760 u69632
[    0.000000] pcpu-alloc: s32680 r8192 d28760 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] CPU with hartid=0 is not available
[    0.000000] riscv-intc: unable to find hart id for /cpus/cpu@0/interrupt-controller
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [1]
[    0.019114] smp: Bringing up secondary CPUs ...
[    0.027611] smp: Brought up 1 node, 4 CPUs

# cat /proc/cpuinfo
processor       : 0
hart            : 1
isa             : rv64imafdc
mmu             : sv39
uarch           : sifive,u54-mc
cpu MHz         : 0.00

processor       : 1
hart            : 2
isa             : rv64imafdc
mmu             : sv39
uarch           : sifive,u54-mc
cpu MHz         : 0.00

processor       : 2
hart            : 3
isa             : rv64imafdc
mmu             : sv39
uarch           : sifive,u54-mc
cpu MHz         : 0.00

processor       : 3
hart            : 4
isa             : rv64imafdc
mmu             : sv39
uarch           : sifive,u54-mc
cpu MHz         : 0.00

# ls -l /sys/devices/system/cpu/
total 0
drwxr-xr-x    5 root     root             0 Jan  1 00:00 cpu0
drwxr-xr-x    5 root     root             0 Jan  1 00:00 cpu1
drwxr-xr-x    5 root     root             0 Jan  1 00:00 cpu2
drwxr-xr-x    5 root     root             0 Jan  1 00:00 cpu3
drwxr-xr-x    2 root     root             0 Jan  1 00:00 hotplug
-r--r--r--    1 root     root          4096 Jan  1 00:00 isolated
-r--r--r--    1 root     root          4096 Jan  1 00:00 kernel_max
-r--r--r--    1 root     root          4096 Jan  1 00:00 offline
-r--r--r--    1 root     root          4096 Jan  1 00:00 online
-r--r--r--    1 root     root          4096 Jan  1 00:00 possible
-r--r--r--    1 root     root          4096 Jan  1 00:00 present
drwxr-xr-x    2 root     root             0 Jan  1 00:00 smt
-rw-r--r--    1 root     root          4096 Jan  1 00:00 uevent

# cat /sys/devices/system/cpu/online
0-3
# cat /sys/devices/system/cpu/offline

# cat /sys/devices/system/cpu/possible
0-3
# cat /sys/devices/system/cpu/present
0-3
# cat /sys/devices/system/cpu/kernel_max
7

```

Thanks for your help,

nearly-big-endian, 2021-12-15:
Hi,

Some updates & precisions:
- my Yocto dev environment was cloned from this github repo on November 17th, so is actually closer to the [2021.11 release](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.11) (released December 10th) than the [2021.08 release](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.08) (released August 6th)
- Unlike stated in my first post, the [2021.08 release](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.08) DOES correctly handle all 4 cpu cores (according to new tests on my icicle-kit-es with the [core-image-minimal-dev-icicle-kit-es-20210806140826.rootfs.wic.gz](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/download/v2021.08/core-image-minimal-dev-icicle-kit-es-20210806140826.rootfs.wic.gz))
- On the other hand the [2021.11 release](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.11) does NOT handle the 4 cpu cores anymore (according to new tests on my icicle-kit-es with the [core-image-minimal-dev-icicle-kit-es-20211208131041.rootfs.wic.gz](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/download/v2021.11/core-image-minimal-dev-icicle-kit-es-20211208131041.rootfs.wic.gz))

So this looks like regression between release [2021.08](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.08) and [2021.11](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2021.11)

lhanlyu, 2021-12-16:
I have retested this core-image-minimal-dev and am not seeing reported issue.
I suggest delete the build directory and rebuild.
> repo sync  (as made some late changes to the defconfig)
> run the setup script
>bitbake <image>

nearly-big-endian, 2021-12-16:
Hello lhanlyu.

Thanks for your answer.

The thing is that I am also seeing the issue with the [latest core-image-minimal-dev-icicle-kit-es image release](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/download/v2021.11/core-image-minimal-dev-icicle-kit-es-20211208131041.rootfs.wic.gz) flashed on a blank SD card.

So that should rule out my custom build.

And the [2021.08 core-image-minimal-dev-icicle-kit-es image](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/download/v2021.11/core-image-minimal-dev-icicle-kit-es-20211208131041.rootfs.wic.gz), when flashed on a blank SD card, does not exhibit the issue. 

Or is it possible that the [reference design](https://github.com/polarfire-soc/icicle-kit-reference-design) must be updated on the board as well ? (my board is currently running HSS 0.99.16).

nearly-big-endian, 2021-12-16:
Okay I updated the board to the latest [reference design](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2021.11) (which includes HSS 0.99.26) and it fixed the problem !

(both with the latest released image and my custom built image)

So it was just a matter of having all components aligned.

Thanks for your time.

closed 2021-12-16 by nearly-big-endian