polarfire-soc/meta-polarfire-soc-yocto-bsp #33: How to get previous tags of meta-polarfire-soc-yocto-bsp project

nearly-big-endian, 2021-12-23:
Hi,

I have a question about release tags.

It seems that, regardless of what release tag I attempt to fetch, I always get the latest (=master) branch of the meta-polarfire-soc-yocto-bsp project.

How should I proceed if I wanted to obtain the actual state of the meta-polarfire-soc-yocto-bsp project as it was when, let's say, release 2021.08 was tagged ?

To illustrate the point, if I fetch the 2021.11 release:
```
$ mkdir yocto-dev-v2021.11
$ cd yocto-dev-v2021.11
$ repo init -u https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp.git -b refs/tags/v2021.11 -m tools/manifests/riscv-yocto.xml
$ repo sync
$ repo rebase
$ repo info
Manifest branch: master
Manifest merge branch: refs/tags/v2021.11
Manifest groups: all,-notdefault
----------------------------
Project: meta-openembedded
Mount path: /home/nagra/yocto-dev-v2021.11/meta-openembedded
Current revision: 77718718cf0eec2c7119d101a353f343e2769b1b
Manifest revision: 77718718cf0eec2c7119d101a353f343e2769b1b
Local Branches: 0
----------------------------
Project: meta-polarfire-soc-yocto-bsp
Mount path: /home/nagra/yocto-dev-v2021.11/meta-polarfire-soc-yocto-bsp
Current revision: 02cb81741a5e1b37f89318d04ebc6b2854bfe736
Manifest revision: master
Local Branches: 0
----------------------------
Project: meta-riscv
Mount path: /home/nagra/yocto-dev-v2021.11/meta-riscv
Current revision: 47faa008327abc388c52246b24a8a458b30b6f0d
Manifest revision: 47faa008327abc388c52246b24a8a458b30b6f0d
Local Branches: 0
----------------------------
Project: openembedded-core
Mount path: /home/nagra/yocto-dev-v2021.11/openembedded-core
Current revision: 63a235d86edb5ca628c8601ed06fb04da3186141
Manifest revision: 63a235d86edb5ca628c8601ed06fb04da3186141
Local Branches: 0
----------------------------
Project: bitbake
Mount path: /home/nagra/yocto-dev-v2021.11/openembedded-core/bitbake
Current revision: 473e2a5486bd972ad0f808db089abcb8945d3a48
Manifest revision: 473e2a5486bd972ad0f808db089abcb8945d3a48
Local Branches: 0
----------------------------
```
I get 'master' (latest) for the manifest revision of project meta-polarfire-soc-yocto-bsp

And if I fetch the 2021.08 release:
```
$ mkdir yocto-dev-v2021.08
$ cd yocto-dev-v2021.08
$ repo init -u https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp.git -b refs/tags/v2021.08 -m tools/manifests/riscv-yocto.xml
$ repo sync
$ repo rebase
$ repo info
Manifest branch: master
Manifest merge branch: refs/tags/v2021.08
Manifest groups: all,-notdefault
----------------------------
Project: meta-openembedded
Mount path: /home/nagra/yocto-dev-v2021.08/meta-openembedded
Current revision: dcb9ecc1e5720c9614b1cd27575e1e4886dff5c1
Manifest revision: dcb9ecc1e5720c9614b1cd27575e1e4886dff5c1
Local Branches: 0
----------------------------
Project: meta-polarfire-soc-yocto-bsp
Mount path: /home/nagra/yocto-dev-v2021.08/meta-polarfire-soc-yocto-bsp
Current revision: 02cb81741a5e1b37f89318d04ebc6b2854bfe736
Manifest revision: master
Local Branches: 0
----------------------------
Project: meta-riscv
Mount path: /home/nagra/yocto-dev-v2021.08/meta-riscv
Current revision: 35d1d5849294ff9f2fb98b90657bbf7a1e532d0d
Manifest revision: 35d1d5849294ff9f2fb98b90657bbf7a1e532d0d
Local Branches: 0
----------------------------
Project: openembedded-core
Mount path: /home/nagra/yocto-dev-v2021.08/openembedded-core
Current revision: 14795ee305f3c11fcc31cc7ca815b8ff1020e29a
Manifest revision: 14795ee305f3c11fcc31cc7ca815b8ff1020e29a
Local Branches: 0
----------------------------
Project: bitbake
Mount path: /home/nagra/yocto-dev-v2021.08/openembedded-core/bitbake
Current revision: 9dc77266085c605b108641a9d76ac4dbdc064c34
Manifest revision: 9dc77266085c605b108641a9d76ac4dbdc064c34
Local Branches: 0
````

... I also get 'master' (latest) for the meta-polarfire-soc-yocto-bsp project.

So I am basically always getting the latest for the meta-polarfire-soc-yocto-bsp project.

How could I get the meta-polarfire-soc-yocto-bsp project revision that was the current at release time ?

Thanks,

vfalanis, 2021-12-23:
Hi,

At the moment, the riscv-yocto.xml manifest and recipes from meta-polarfire-soc will automatically fetch the last commit from main/master branch. This is something that will be addressed in our next release.

In the meantime, you could use [buildroot](https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk/tree/v2021.08) to build an image from a previous release (i.e. tag v2021.08)

lhanlyu, 2022-03-10:
Branch created for each previous release, use these branch names with repo init. Please see the readme on each branch for more details.

closed 2022-03-10 by lhanlyu