polarfire-soc/meta-polarfire-soc-yocto-bsp #34: git:// uri deprecated

notthat9uy, 2022-01-11:
Looks like github.com no longer supports git:// uris. I've truncated the output below

```
repo sync

fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.

meta-openembedded:
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
meta-openembedded: sleeping 4.0 seconds before retrying
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.

meta-openembedded:
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
error: Cannot fetch meta-openembedded from git://github.com/openembedded/meta-openembedded
Fetching: 20% (1/5) meta-openembedded

Fetching: 40% (2/5) meta-polarfire-soc-yocto-bsp
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.

meta-riscv:
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
meta-riscv: sleeping 4.0 seconds before retrying
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.                                                                                                                                                                                                                                              
```

Manually modifying the uris with https in .repo manifest after `repo init -u https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp.git -b master -m tools/manifests/riscv-yocto.xml` gets you a bit further.

There are git uris in the project:
```
recipes-amp/mpfs-amp-examples/mpfs-amp-examples.bb:SRC_URI = "git://github.com/polarfire-soc/polarfire-soc-amp-examples.git;branch=${BRANCH}"
tools/manifests/riscv-yocto.xml:  <remote name="openembedded" fetch="git://github.com/openembedded/" />
tools/manifests/riscv-yocto.xml:  <remote name="riscv" fetch="git://github.com/riscv/" />
tools/manifests/bb_develop.xml:  <remote name="openembedded" fetch="git://github.com/openembedded/" />
tools/manifests/bb_develop.xml:  <remote name="riscv" fetch="git://github.com/riscv/" />
tools/manifests/bb_testb.xml:  <remote name="openembedded" fetch="git://github.com/openembedded/" />
tools/manifests/bb_testb.xml:  <remote name="riscv" fetch="git://github.com/riscv/" />
recipes-bsp/hss/hss.bb:SRC_URI = "git://github.com/polarfire-soc/hart-software-services.git;branch=${BRANCH} \
recipes-tools/coremarkpro/coremarkpro.bb:SRC_URI = "git://github.com/eembc/coremark-pro.git;branch=main"
recipes-tools/pfsoc_apps/pfsocapps.bb:SRC_URI = "git://github.com/polarfire-soc/polarfire-soc-linux-examples.git;branch=${BRANCH} \
recipes-kernel/linux/mpfs-linux_5.%.bb:    git://github.com/polarfire-soc/linux.git;branch=${BRANCH} \
```

Changing these to https got me further but I still ran into other build issues
```
MACHINE=icicle-kit-es bitbake mpfs-dev-cli
Loading cache: 100% |################################################################################################################################################################################################################| Time: 0:00:02
Loaded 3765 entries from dependency cache.
WARNING: /home/pi/polarfire/meta-polarfire-soc-yocto-bsp/yocto-dev/meta-polarfire-soc-yocto-bsp/recipes-kernel/linux/mpfs-linux_5.%.bb: Unable to get checksum for mpfs-linux SRC_URI entry defconfig: file could not be found      | ETA:  --:--:--
Parsing recipes: 100% |##############################################################################################################################################################################################################| Time: 0:00:01
Parsing of 2428 .bb files complete (2425 cached, 3 parsed). 3768 targets, 177 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES ''vmlinux',' (but /home/pi/polarfire/meta-polarfire-soc-yocto-bsp/yocto-dev/meta-polarfire-soc-yocto-bsp/recipes-kernel/linux/mpfs-linux_5.%.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target ''vmlinux',' is unbuildable, removing...
Missing or unbuildable dependency chain was: ["'vmlinux',"]
ERROR: Required build target 'mpfs-dev-cli' has no buildable providers.
Missing or unbuildable dependency chain was: ['mpfs-dev-cli', 'virtual/kernel', "'vmlinux',"]

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

This is where I stopped trying to modify the project to get the build working

lhanlyu, 2022-01-12:
The fetch protocol will be updated in the next release.

lhanlyu, 2022-03-10:
https protocol supported for this BSP.

closed 2022-03-10 by lhanlyu