polarfire-soc/meta-polarfire-soc-yocto-bsp #36: u-boot version 2020.01

kowalski100, 2022-02-01:
Hi,

Our Icicle board contains u-boot version 2020.01. The .wic images shared on this repo contains later version.

Is there a link pointing to above version?

vfalanis, 2022-02-01:
Hi,

The following binaries/assets can be downloaded from our repos:

1. Pre-built WIC images per release from the [release page](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases)
2. Job files per release from the Icicle Kit reference design [release page](https://github.com/polarfire-soc/icicle-kit-reference-design/releases)

If required, it is possible to extract the U-boot and a fitImage (Linux + DTB) from a WIC image from a particular release. For an example on how to extract U-Boot from a WIC file please refer to this [link](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/issues/35)

The table below contains the versions of the binaries used in some of our most recent releases:
| PolarFire SoC Yocto Release | U-Boot  | Linux   | HSS      | Reference Design |
|-----------------------------|---------|---------|----------|------------------|
| v2021.11                    | 2021.07 | 5.12.19 | v0.99.26 | v2021.11         |
| v2021.08                    | 2021.04 | 5.12.1  | v0.99.23 | v2021.08         |
| v2021.04                    | 2020.1  | 5.6.18  | v0.99.16 | v2021.04         |
| v2021.02                    | 2020.1  | 5.6.18  | v0.99.15 | v2021.02         |
| v2020.11                    | 2020.07 | 5.3.18  | v0.99.14 | v2020.11         |

kowalski100, 2022-02-01:
Hi,

None of the above mentioned releases contains default images on the icicle board. As I mentioned in the thread, the default u-boot version prints 2020.01. We want the board to get back to default state and for that default binaries are needed.

vfalanis, 2022-02-04:
Hi,

You can get the default Icicle Kit image and HSS job file from this [link](https://microchiptechnology-my.sharepoint.com/:u:/g/personal/nitin_deshpande_microchip_com/EZXwnT605FFLjBUdbmmtNFQBS6g9V6OuqbVX6yWuQDGB1g?e=rteCxG).