polarfire-soc/meta-polarfire-soc-yocto-bsp #38: Not all available memory usable

mastercaution, 2022-02-16:
Hi,

I'm running the pre-build image for release 2021.11 (` core-image-minimal-dev-icicle-kit-es-20211208131041.rootfs.wic.gz `) on my board, which was also updated to the Reference Design 2021.11.

For some reason, I have less than 1 GB RAM available since I've updated:
```
root@icicle-kit-es:~# free
               total        used        free      shared  buff/cache   available
Mem:          811344       39524      725888        9628       45932      748220
Swap:        1952232           0     1952232
```

Looks like the kernel can only use DDR-Lo but not DDR-Hi, although it is detected by the HSS:
```
[4.22658] HSS_DDRPrintSegConfig(): Segment Configuration:
        Cached: SEG0_0: offset 0x0080000000, physical DDR 0x00000000
        Cached: SEG0_1: offset 0x1000000000, physical DDR 0x30000000
    Non-cached: SEG1_2: offset 0x00c0000000, physical DDR 0x70000000
Non-cached WCB: SEG1_4: offset 0x00d0000000, physical DDR 0x70000000
[4.53905] HSS_DDRPrintL2CacheWaysConfig(): L2 Cache Configuration:
    L2-Scratchpad:  4 ways (512 KiB)
         L2-Cache:  8 ways (1024 KiB)
           L2-LIM:  4 ways (512 KiB)
[4.71613] HSS_MemTestDDRFast(): DDR-Lo size is 768 MiB
[4.78036] HSS_MemTestDDRFast(): DDR-Hi size is 1 GiB
```

lhanlyu, 2022-02-18:
This backlog issue fixed and will be working correctly in the next release 2022.02, which should be available next week.

mastercaution, 2022-03-08:
Thank you, v2022.02 fixed it!

closed 2022-03-08 by mastercaution