polarfire-soc/meta-polarfire-soc-yocto-bsp #39: Bitbake Fetcher Error while building qemu

balaji-ch, 2022-02-17:
**Build instructions used**
```
mkdir yocto-dev && cd yocto-dev
repo init -u https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp.git -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo rebase
source ./meta-polarfire-soc-yocto-bsp/polarfire-soc_yocto_setup.sh
cd ..
source ./meta-riscv/setup.sh    # without this I get Nothing PROVIDES 'opensbi'
MACHINE=qemuriscv64 bitbake core-image-minimal-dev
```

**ERROR**
ERROR: u-boot-1_2021.07-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'file://.txt')

log attached

[log.do_fetch.txt](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/files/8086511/log.do_fetch.txt)

vfalanis, 2022-02-18:
Hi,

At the moment there's no support for qemu machines in PolarFire SoC Yocto BSP. This is something we will update in the future.

We'll update the README with the current list of compatible machines

lhanlyu, 2022-05-04:
Not supporting qemu with this BSP.

closed 2022-05-04 by lhanlyu