polarfire-soc/meta-polarfire-soc-yocto-bsp #48: Kernel panic - not syncing: Attempted to kill the idle task!

Jpgig, 2023-04-05:
While running polarfore-soc-yocto-bsp 2021.11 version from the below link:
https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/tree/2021.11

I'm getting Kernel panic error as shown in below screen shot.
[image]

Appreciate help to resolve this.

lhanlyu, 2023-04-05:
This issue was fixed in later releases. can you try the latest release.

Jpgig, 2023-04-05:
I want to integrate ROS Honister release so need to use this branch - Honister. Can you suggest the modification?

lhanlyu, 2023-04-07:
Can you provide more logs? Later releases also use Honister?

Jpgig, 2023-04-10:
Below are more details about the log:
[image]

[image]

Also, where can I find the mapping of the reference design to the Polarfire yocto build version? I am using Polarfire soc yocto build the 2021.11 version.

lhanlyu, 2023-05-30:
The icicle-kit reference design and yocto bsp releases are always on the same release date.
If using the 2021.11 yocto release, you should use  https://github.com/polarfire-soc/icicle-kit-reference-design/releases 2021.11

If at all possible you should try to update.

closed 2023-07-27 by lhanlyu