polarfire-soc/meta-polarfire-soc-yocto-bsp #56: Instructions to boot mpfs-disco-kit from SD card  (attempted and failed)

NavadeepGaneshU, 2024-06-06:
Hi, 
I am trying to boot the latest released image from (https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2024.02.1) on the MPFS Disco Kit and after inserting the SD card, nothing is showing up on the serial terminal. Is there any additional consideration for this kit?
Some points:
1. Flashed the core-image-minimal-dev-mpfs-disco-kit-20240426091646.rootfs.wic file using windows disk imager and it showed 3 sections inside the SD card.
2. Hot plugged the SD card while powered ON and connected to UART. Tried the other way as well my connecting after inserting the SD card.
3. I had programmed the (https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/blob/main/driver-examples/mss/mss-gpio/mpfs-gpio-interrupt) example before to the board and it this was working. Now when I connect the serial terminal, the serial output of this example with ' Microchip MPFS-DISCO-KIT Test' is getting printed in COM9 (among COM8/9/10). I suppose it is from eNVM and should this be cleared for linux to boot?
4. I have tried loading my own yocto build run on WSLv2 and the reference eample image given at (https://github.com/polarfire-soc/polarfire-soc-discovery-kit-reference-design/releases/tag/2024.04) as well from any of which I couldn't find the UART terminal showing up with the U-Boot loading.

hughbreslin, 2024-06-06:
Hi @NavadeepGaneshU it sounds like you programmed the GPIO example into the eNVM? This has probably overwritten the HSS which is used as a boot loader for Linux. Can you try reprogramming the discovery kit reference design using FPExpress (this includes the HSS) and see if that helps your issue.

NavadeepGaneshU, 2024-06-07:
Thanks for the quick response @hughbreslin. Yes, I was doing that with the GPIO example and after programming the .job file from the reference design example, it worked out. 
Next, trying to control a GPIO pin (GPIO2, PIN17) and I am having gpio@20120000, gpio@20121000, gpio@20122000 identified inside the `device-tree/soc` . Can you point me to a reasonable reference or guide how to configure this pin as GPIO and command it to write 1 and 0?
(I see mpfs-icicle-kit comes loaded with gpiotest example)

NavadeepGaneshU, 2024-06-09:
Thanks for the detailed explanation. "Ask Hugh" is a pretty great phrase. Now I can use python script to toggle and control them in any order. Adding onto this, I tried extending the device-tree overlay for I2C. Knowing there is libi2cd present and I am seeing `i2c@2010a000` , `i2c@2010b000` etc, I included
`
 i2c0: i2c@2010a000 {
            compatible = "microchip,mpfs-i2c";
            reg = <0x00 0x2010a000 0x00 0x1000>;
            interrupt-parent = <&irqmux>;
            interrupt-controller;
            clock-frequency = <400000>;
            i2c-controller;
            #address-cells = <1>;
            #size-cells = <0>;
            status = "okay";

            htu21d@40 {
                compatible = "te,htu21d";
                reg = <0x40>;
                status = "okay";
            };
        };
`

but the device i2c peripheral doesn't seem to appear under `ls /dev/`. Sorry I know I am going beyond what is asked in the original question, but a clarification on this will be of help in kit review and development I am doing.

hughbreslin, 2024-06-09:
Haha great 👍 if you want you can add a script to automatically apply the overlay at boot so you don’t have to do it manually as well on each startup :)

Looking at what you shared for I2C, MSS I2C_0 is enabled on the kit. I had a look at our base device tree entry [here](https://github.com/linux4microchip/linux/blob/99b20cea90ff1a05163da777cb95b0dae2016ef1/arch/riscv/boot/dts/microchip/mpfs.dtsi#L381) and I can see a few differences: 

`
		i2c0: i2c@2010a000 {
			compatible = "microchip,mpfs-i2c", "microchip,corei2c-rtl-v7";
			reg = <0x0 0x2010a000 0x0 0x1000>;
			#address-cells = <1>;
			#size-cells = <0>;
			interrupt-parent = <&plic>;
			interrupts = <58>;
			clocks = <&clkcfg CLK_I2C0>;
			clock-frequency = <100000>;
			status = "disabled";
		};
`

I’ll be able to try it out tomorrow but I’d say get most of it lined up (and marked enabled!) and fingers crossed it should work :)

hughbreslin, 2024-06-10:
Hey @NavadeepGaneshU 

I tried out the following:

```
			i2c0: i2c@2010a000 { 
				compatible = "microchip,mpfs-i2c", "microchip,corei2c-rtl-v7";
				reg = <0x0 0x2010a000 0x0 0x1000>;
				#address-cells = <1>;
				#size-cells = <0>;
				interrupt-parent = <&plic>;
				interrupts = <58>;
				clocks = <0x01 0xF>;
				clock-frequency = <100000>;
				status = "okay";
			};
```

And I can see I2C-0 show up in /dev/i2c* :) I don't have anything to test it with but hopefully this is enough to get you on the right track. I got the clock value from the [mpfs-clock.h file](https://github.com/linux4microchip/linux/blob/99b20cea90ff1a05163da777cb95b0dae2016ef1/include/dt-bindings/clock/microchip%2Cmpfs-clock.h#L26C18-L26C20).

This thread has actually been quite useful and has gone beyond the scope a bit but could be a useful discussion item so its more public. I'll get it converted over :)

NavadeepGaneshU, 2024-06-11:
Thanks for following up @hughbreslin. The device gets added to the `/dev/` list now as you've directed while the reference design image is loaded. I was on the custom yocto build image made using the instructions in this repo. Having trouble in getting the correct I2C sensor address detected by the `i2detect 0` search but it can be on my side.

closed 2024-06-11 by NavadeepGaneshU