polarfire-soc/meta-polarfire-soc-yocto-bsp #57: mmap on reserved-memory blocks the program

Ryose, 2024-06-11:
Hi, I'm using the Yocto-BSP for the enclustra-mpfs SoM wich mounts a PolarFire SoC device. 
Hi have a node in my device-tree which is used to reserve a set of cached memory: 
```
/{
    reserved-memory {
    #address-cells = <2>;
    #size-cells = <2>;
[mem-prova.zip](https://github.com/user-attachments/files/15788146/mem-prova.zip)
    ranges;
    
        reserved: buffer@0 {
            compatible = "shared-dma-pool";
            no-map;
            reg = <0x0 0xC0000000 0x0 0x10000000>;
        };
    };
    
    reserved-driver@0 {
    compatible = "ingeniars,gpu4sat-memory";
    memory-region = <&reserved>;
    };
};
```
My kernel module uses the DMA API to reserve this memory area using a dma_alloc_coherent() and performs a mmap() to give access to a user application, which I attached. Unfortunately, my application blocks indefinitely. 
[mem-prova.zip](https://github.com/user-attachments/files/15788205/mem-prova.zip)

I also attach the mss memory partition configuration of my project and the output of the execution of my application. 
Thanks for your help in advance.
[image]
[image]

eppidei, 2024-06-13:
Hi @Ryose , If I understand well the memory gets allocated but the DMA operation stops. Is that correct?  Have you specified the DMA ranges in the device tree?

Ryose, 2024-06-14:
Yes, I thinks that the allocation is performed correctly( since no error is generated) but when I access said memory region the process stalls.  As you can see, I have declared the node of reserved memory inside my device-tree, is there any additional node that I need to add to my device-tree?