polarfire-soc/meta-polarfire-soc-yocto-bsp #63: Understand the Boot Up sequence

ankrtlpixxel, 2025-04-22:
Hello,
I am using the Microchip Discovery Kit and would like to understand the boot-up sequence for the device.

Here is what I have done:

- Built a custom FPGA design using the Libero tool, based on the pre-existing FPGA block provided in the YOCTO-BSP.
- Modified the Yocto BSP to update the Device Tree Source (mpfs-disco-kit-fabric.dtsi) and added the following nodes so that Linux can recognize the new devices:

```
	//UART0 
	uart5: serial@43000000 {
		compatible = "ns16550a";
		reg = <0x0 0x43000000 0x0 0x400>;
		reg-io-width = <4>;
		reg-shift = <2>;
		interrupt-parent = <&plic>;
		interrupts = <123>;
		clocks = <&ccc_nw CLK_CCC_PLL0_OUT3>;
		current-speed = <115200>;
		status = "okay";
	};
```

         //UART1
		uart6: serial@40000400 {
			compatible = "ns16550a";
			reg = <0x0 0x20106000 0x0 0x400>;
			reg-io-width = <4>;
			reg-shift = <2>;
			interrupt-parent = <&plic>;
			interrupts = <121>;
			clocks = <&ccc_nw CLK_CCC_PLL0_OUT3>;
			current-speed = <115200>;
			status = "okay";
		};
'
- Added a meta-pixxel layer to modify the NR_UARTS kernel configuration from the default value of 2 to 8.

The **_.wic_** file generated by building mpfs-dev-cli is transferred to an SD card, and the device boots successfully. However, when I flash the new FPGA bitstream, the device fails to boot.

Questions:
1.) Is the FPGA flashing process correct?
2.) Is there a standard way to include the FPGA bitstream as part of the SD card image?
3.) Can the device tree be modified in the way described above?

0xEnsar, 2025-04-26:
Did you tried your FPGA design with bare-metal applications ? Is your HSS updated ?  @ankrtlpixxel

ankrtlpixxel, 2025-10-30:
Got this running  after modifying the HSS code , using the updated MSS cxz file.

closed 2025-10-30 by ankrtlpixxel